pvaret/rtl8192cu-fixes #130: Forked/Updated for kernel 4.15.0

rburcham, 2018-01-31:
You can find it here:

https://github.com/rburcham/rtl8192cu-fixes

KinkyDragoness, 2018-01-31:
Awesome! Excellent work, thanks. :)

If you'd like to send a PR for commit 313eaccb, I'll gladly merge it. But if you want your repo to become the reference for kernel 4.15+, that's fine with me too.

pvaret, 2018-01-31:
Awesome! Excellent work, thanks. :)

If you'd like to send a PR for commit 313eaccb, I'll gladly merge it. But if you want your repo to become the reference for kernel 4.15+, that's fine with me too.

rburcham, 2018-02-01:
Hah, I wouldn't call a bloody mess of #ifdefs "excellent," but thanks! :)  Submitted the PR, details therein.  You'll likely want to review and re-edit the README after merging.

pvaret, 2018-02-02:
Thanks! PR-related comments on the PR itself.

Do note, I'm glad that you think this highly of this driver, but there are sadly tons of devices that don't work well or even at all with it. Our issue tracker is full of such reports that we unfortunately won't be fixing; this is really Realtek's own driver and we're not maintaining it, we're only making it compile on more recent kernels. We point people to the rtl8xxxu one not because it's a silver bullet — it is not — but because it's actively maintained, and actually supports a large number of devices, even if imperfectly. And it only gets better as people use it and report issues.

That said, the fact that this driver may outperform rtl8xxxu when it does work is indeed noteworthy, and I'll add a note in the README to that extent. Thanks for the suggestion!

rburcham, 2018-02-02:
That makes sense.  And FWIW I have forwarded observations and logs to Jes Sorenson re: the dongles I have and their behaviors with  rtl8xxxu.  Will close this and take it up in the PR

closed 2018-02-02 by rburcham