pvaret/rtl8192cu-fixes #131: Blacklist rtl8192cu in order for rtl8xxxu to work.

Pant, 2018-02-01:
Please add a line in README.md in order to explain that someone needs to blacklist `rtl8192cu` in order for `rtl8xxxu` driver to work. I had to do that on Fedora 27 x64, Kernel: 4.14.14-300.fc27.x86_64, and it works good.

`sudo nano /etc/modprobe.d/rtl8192cu.conf`
```
install rtl8192cu /bin/false
```

dlech, 2018-02-01:
It already says:

> Make sure to blacklist the older `rtl8192cu` driver, which tends to be loaded by default otherwise.

rburcham, 2018-02-01:
You are wise to blacklist both rtl8xxxu and rtl8192cu in order to ensure the driver from this github project, 8192cu, can load.

pvaret, 2018-02-01:
As dlech pointed out, this is a good idea, and already done. :)

closed 2018-02-01 by pvaret