pvaret/rtl8192cu-fixes #140: after wake up from suspend no connect to the access point posible

maxpower18, 2018-05-01:
the driver works fine until the version of Feb 3, 2018.
driver version afterwards make problem after wake up from suspend, is no longer posible to connect the access point, see log attachment.
only a computer restart helps.
operating system is ubuntu 16.04 whit kernel 4.13

```
wpa_supplicant[909]: wlan0: Trying to associate with 07:96:d6:44:1f:bc (SSID='FRITZ' freq=2412 MHz)
wpa_supplicant[909]: wlan0: Association request to the driver failed
NetworkManager[708]: <info>  [1525172366.4501] device (wlan0): supplicant interface state: scanning -> disconnected
NetworkManager[708]: <info>  [1525172366.9518] device (wlan0): supplicant interface state: disconnected -> scanning
wpa_supplicant[909]: wlan0: Trying to associate with 07:96:d6:44:1f:bc (SSID='FRITZ' freq=2412 MHz)
wpa_supplicant[909]: wlan0: Association request to the driver failed
NetworkManager[708]: <info>  [1525172368.3141] device (wlan0): supplicant interface state: scanning -> disconnected
NetworkManager[708]: <info>  [1525172369.3156] device (wlan0): supplicant interface state: disconnected -> scanning
wpa_supplicant[909]: wlan0: Trying to associate with 07:96:d6:44:1f:bc (SSID='FRITZ' freq=2412 MHz)
wpa_supplicant[909]: wlan0: Association request to the driver failed
wpa_supplicant[909]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="FRITZ" auth_failures=1 duration=10 reason=CONN_FAILED
NetworkManager[708]: <info>  [1525172370.6701] device (wlan0): supplicant interface state: scanning -> disconnected
NetworkManager[708]: <info>  [1525172376.1046] device (wlan0): supplicant interface state: disconnected -> scanning
wpa_supplicant[909]: wlan0: CTRL-EVENT-SSID-REENABLED id=0 ssid="FRITZ"
wpa_supplicant[909]: wlan0: Trying to associate with 07:96:d6:44:1f:bc (SSID='FRITZ' freq=2412 MHz)
wpa_supplicant[909]: wlan0: Association request to the driver failed
wpa_supplicant[909]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="FRITZ" auth_failures=2 duration=20 reason=CONN_FAILED
NetworkManager[708]: <info>  [1525172384.2706] device (wlan0): supplicant interface state: scanning -> disconnected
....
```

pavel-demin, 2018-05-01:
For information. Here is a link to a similar issue with the rtl8723bu driver:

https://github.com/lwfinger/rtl8723bu/issues/91

pavel-demin, 2018-05-01:
I've looked at how the problem was fixed in the rtl8723bu driver and I think that replacing the lines 724-733 in [os_dep/linux/ioctl_cfg80211.c](https://github.com/pvaret/rtl8192cu-fixes/blob/master/os_dep/linux/ioctl_cfg80211.c#L724-L733) with the [corresponding lines from the rtl8723bu driver](https://github.com/lwfinger/rtl8723bu/blob/master/os_dep/ioctl_cfg80211.c#L835-L844) should fix this problem in the rtl8192cu-fixes driver.

pavel-demin, 2018-05-02:
Thanks for testing the fix. Could you please try to change two more lines in [os_dep/linux/ioctl_cfg80211.c](https://github.com/pvaret/rtl8192cu-fixes/blob/master/os_dep/linux/ioctl_cfg80211.c)?

Here are the changes:
 - comment out `if(check_fwstate(&padapter->mlmepriv, _FW_LINKED)) ` on [line 2928](https://github.com/pvaret/rtl8192cu-fixes/blob/master/os_dep/linux/ioctl_cfg80211.c#L2928)
 - add `rtw_pwr_wakeup(padapter);` just after `rtw_free_assoc_resources(padapter, 1);` on [line 2940](https://github.com/pvaret/rtl8192cu-fixes/blob/master/os_dep/linux/ioctl_cfg80211.c#L2940)

The idea of these changes is to make `cfg80211_rtw_disconnect` look similar to [its version in the rtl8723bu driver](https://github.com/lwfinger/rtl8723bu/blob/master/os_dep/ioctl_cfg80211.c#L3103-L3132).

pavel-demin, 2018-05-03:
While attempting to reproduce the problem, I've installed Ubuntu 18.04 with Linux kernel 4.15.0-20. I've tried a few times to suspend and to wake up the system and my WiFi connection always gets reestablished after the wake up.

maxpower18, 2018-05-03:
I have now installed kernel 4.15.18 under ubuntu 16.04 and tried the driver with and without code changes but the problem is still there.
So it seems not only the kernel maybe new version of network manager or something that it works in ubuntu 18.04.

pavel-demin, 2018-05-03:
I've just installed Ubuntu 16.04.4, Linux kernel 4.13.0-39-generic (via the `linux-generic-hwe-16.04` package) and Network Manager 1.2.6. I'm using `systemctl suspend` to suspend the system and the power button to wake it up. Still can't reproduce the problem. I don't have the `Association request to the driver failed` messages in the logs.

maxpower18, 2018-05-03:
I have run exactly the same setup, to wake up I move the mouse but that should make no difference.
So only wifi hardware remains, I use EDIMAX EW-7811UN usb wifi sick on usb 2.0.
lsusb:
`Bus 003 Device 003: ID 7392: 7811 Edimax Technology Co., Ltd. EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]`

pavel-demin, 2018-05-03:
I have TP-Link TL-WN725N:
```
ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
```

pavel-demin, 2018-05-03:
Could you please check the command line options of the `wpa_supplicant` process? Is it started with `-D wext` or without?

maxpower18, 2018-05-03:
> Could you please check the command line options of the wpa_supplicant process?

How can I do that?

maxpower18, 2018-05-03:
```
root       912  0.0  0.1  44028  7212 ?        Ss   10:23   0:00 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant
max      13470  0.0  0.0  15796   940 pts/1    S+   15:09   0:00 grep --color=auto wpa_supplicant

```

dsd, 2019-07-03:
Try editing `os_dep/linux/os_intfs.c`
Find the `netdev_close` function.

Find the line `#ifndef CONFIG_ANDROID`
Remove that line, and the `#endif` that follows soon after.