pvaret/rtl8192cu-fixes #142: 4.15 fix does not work well

Galcian79, 2018-05-30:
I have a general brand wifi adapter with rtl8192cu chip.
At first i installed this driver on Ubuntu 17.10, not satisfied by the open source one, and it worked fine. Now i switched to 18.04 and applied the 4.15 fix, but it's not working well: weak signal strength and data speed loss. It still works fine on 4.13 though.
Now the 4.13 kernel support will end in july and i don't like the idea to stick to an old unsupported kernel due to this problem.
I wonder if there is a way to improve this driver further or, at least, get instructions on how to restore the open source one.
Thanks in advance.

CGarces, 2018-05-30:
Consider use the rtl8xxxu that is included in the kernel since 4.4
You can also try the kernel module rtl8192cu but It's going to be replaced by rtl8xxxu.

Galcian79, 2018-06-03:
So in /etc/modprobe.d/blacklist-native-rtl8192.conf i just comment the line`install rtl8xxxu /bin/false`

closed 2018-06-03 by Galcian79