pvaret/rtl8192cu-fixes #153: Use only antenna B for 8192cu

joseamartins, 2019-02-28:
Hello pvaret,

I am not sure if here is the appropriated location to clarify doubts. If not, sorry.

I'd like to configure the 8192cu to use just one antenna (1T1R). Is there a way to do that?

I have a board with this device and I have mounted just antenna B. Using the mainline rtl8192cu driver, I can scan a lot of networks with good quality signal.

When I use 8192cu driver, I can list just near networks, with poor signal. I think the reason is that the 8192 tries to use both antennas.

Thanks in advance.

Jose M.

pvaret, 2019-03-11:
Hi Jose,

Thank you for your confidence, but I should probably state it again: this repo is only a pile of hacks to make Realtek's own driver compile on non-antiquated kernels. We don't actually maintain it or add features. In a way, it's a miracle that it still works at all. :)

If you are feeling very optimistic, maybe you'll want to take your question to Realtek themselves?

Meanwhile, if the mainline driver actually works for you, I can only recommend that you use that, as that one is actually maintained. Sorry I can't help you beyond that!

closed 2019-03-11 by pvaret

joseamartins, 2019-03-12:
Hi pvaret,

Thanks for your replay :)

In fact, the mainline driver works, but not satisfactorily. There is a high cpu consumption in some situations, and even it hangs up other times. And with rtl8192cu-fixes driver I never saw it happen.

Well, I'll try to contact Realtek.

Again, thank you.

José