pvaret/rtl8192cu-fixes #155: Can't connect to the wifi with wicd after installing this driver - BAD PASSWORD.

KoStard, 2019-04-07:
I'm getting so slow connection with default drivers, so I try to fix that... I have installed wicd, but that didn't change anything... Now I have installed this and can't connect to the wifi at all... I'm getting "bad password" error from wicd.
So what can I do? Any workaround?

KoStard, 2019-04-07:
Hmm... The problems is solved, even don't understand how... I have installed this driver, activated, got problem with password, deactivated it and again activated the 8xxxu driver and it is now working like a charm... I was getting so slow wifi with 8xxxu driver, that's why I was trying to use this one... So is it possible that the installation of this driver fixed the default one? Because I got more than 40x increase in the wifi speed.

KoStard, 2019-04-07:
Actually, the problem is back after some reboots... So any suggestions?

KoStard, 2019-04-07:
But when I activate this driver, reboot, deactivate and reactive rtl8xxxu, I'm again getting high-speed wifi connection.

pvaret, 2019-04-07:
This repo only contains a repackaging of Realtek's own driver, with patches to make it compile against recent kernels. I don't debug or otherwise maintain the driver itself. You can reach out to Realtek about your problems, but all told it would be much more useful to reach out to the maintainers of the mainline rtl8xxxu driver about your issues so they can fix them, to the ultimate benefit of everyone.

closed 2019-04-07 by pvaret