pvaret/rtl8192cu-fixes #161: This is still the best driver for my device, and yet is depricated.

draxil, 2020-07-01:
I know an issue may not be the best way to communicate this, and feel free to close it; however this driver remains the only one which will run my TL-WN822N properly and remains vastly superior to the rtl8xxxu driver.

The reason this is a problem is that I feel people will see t hat this driver is deprecated and the rtl8xxxu is recommended as a replacement. For my device it claims to work, connects very slowly and with no LED. Where as this driver remains excellent.

For anyone out there struggling with a device like mine I'd urge them to still try this driver, despite it being deprecated.

CGarces, 2020-07-02:
Hi

Deprecated means not maintained any more. It just works for you on your current configuration, but it will change in the future, for example not compile anymore on kernel 5.8 due recent changes.
https://travis-ci.org/github/CGarces/rtl8192cu-fixes/jobs/704249958

Consider open a well documented bug report on your distribution about your problems with TL-WN822N.

lautaromss, 2020-07-04:
@CGarces I just wanted to say that I have the same TL-WN822N and under the latest Ubuntu 20.4 your stuff in this repository remains the ONLY way to make it work.

I don't know what's up with that rtl8xxxu; my OS seemed to come with a generic rtl8192cu and I followed a ton of tutorials about installing some "8192eu" driver that never worked.

After hours and hours of not making it work, I saw another person in askubuntu saying that he tried this repo _despite_ the deprecation warning so I gave it a shot. Typed your stupid commands and now it works like a charm. I have 100x more speed.

I think if you put a message in the README.md clarifying that deprecated or not it might be the best solution specially for TL-WN822N you might save a lot of people hours of trying other stuff.

MiChaelinzo, 2020-07-20:
i have tl-wn823n v2 works well for me currently using it in kali linux monitoring and sniffing