pvaret/rtl8192cu-fixes #165: driver not working for raspbian kernel 5.4.51

MrStashley, 2020-07-25:
I am trying to use this driver to make use of private ioctls for wifi p2p with a raspberry pi and an edimax ew7811un. When following the install instructions, all of the commands run as if they are working correctly, but once I copy the blacklist-native-drivers.conf and reboot, the driver stops working completely; it isn't showing up in lsmod and the internet isn't working. Can anyone help me with this, or is my platform not supported? I really hope I can get this to work because it seems like this driver is the only one that might work on my raspberry pi.

gps05, 2020-08-14:
You could try deleting all lines in the blacklist-native-drivers.conf file and adding `blacklist rtl8192cu`
Or maybe `install rtl8192cu /bin/false`

MrStashley, 2020-08-14:
okay thanks! I'll try it. 
Would install rtl8192cu /bin/false also go inside blacklist-native-drivers.conf?

MrStashley, 2020-08-15:
IT WORKS!!!!!!!
Thank you so much, I think adding 8192cu to /etc/modules made it run. Also there was a blacklist-8192cu.conf in /etc/modprobe.d for some reason that said `blacklist 8192cu`. I think it was accidentally blacklisting the good drivers, I commented it out with # and rebooted and it worked. I now can use wifi P2P and continue working on my project thanks!

MrStashley, 2020-08-15:
if I close the issue does that mean no one else can comment? What if someone has a similar issue and wants my guidance on what I did to fix mine?

dlech, 2020-08-15:
anyone can still comment on a closed issue

MrStashley, 2020-08-17:
Awesome, thanks!

closed 2020-08-17 by MrStashley